unitreerobotics/unitree_legged_sdk #2: Request binary library compiled for Mac M1 arm

erwincoumans, 2020-12-08:
The new Apple Mac Mini with arm64 CPU is an attractive platform for onboard compute: it is very fast (often faster than high-end AMD Ryzen and Intel) and low-power.

Would it be possible to recompile this library for this platform?

Zhaiweiwei0, 2021-07-14:
Hello, we haven't done any tests for this platform yet, Thank you for your suggestion!

TrivasZhang, 2021-10-08:
yxyang told me he solved it in another way, so I close this issue.
Feel free to reopen it if needed.

Marco-Delgado, 2021-11-23:
> yxyang told me he solved it in another way, so I close this issue. Feel free to reopen it if needed.

Is there any resource on how to compile the legged sdk for m1 macs? I'm attempting to work through it now however I am at a roadblock and unsure how to bypass it.

closed 2022-06-24 by TrivasZhang