unitreerobotics/unitree_legged_sdk #15: send high-level commands (walk), and receive low-level sensor readings

talhen123, 2021-11-09:
Hello, I purchased A1 a few months ago for academic research on Estimation. To do this, I need to send high-level commands (walk), and receive low-level sensor readings (for the Estimation). How can this be done? 
I can not do that under the same portal. Is it possible to work with two portals at the same time?

TrivasZhang, 2021-11-24:
When under high-level control, I'm trying to return low-level date.
Did joint position enough? Will the odometry estimator need joint velocity or others?

itaouil, 2021-11-24:
Hi @TrivasZhang,

If you could include the whole **MotorState.msg** in the **HighState.msg** would be great.

Because for publishing the TF tree I require the joint positions, their velocity and the estimated torque for every single servo.

evyatarluv, 2021-11-25:
Hi @TrivasZhang ,

Can you also add the motor state to sdk v3.3 for A1?

talhen123, 2021-11-25:
Hi @TrivasZhang ,
For the odometry estimator I will need the unfilter(raw) joints position, speed, and torques.
If it is possible to add for SDK 3.3 (I use unitree A1) the whole **MotorState** it would be great and Easier.
tanks!

TrivasZhang, 2021-11-25:
> Hi @TrivasZhang ,
> 
> Can you also add the motor state to sdk v3.3 for A1?

Maybe later for A1 sport mode update. And after this update, some other functions, like vision following will not take effect, only sport mode works. 
Will that ok?

TrivasZhang, 2021-11-25:
Hi @talhen123 
I couldn't get the point why need **raw** date for  joints position, speed, and torques. 
That will change the whole communication related codes and seems take little effect.

talhen123, 2021-11-25:
Hi @TrivasZhang 
I need the raw data only for the joints position and velocity to calculate the measurement noise for the kalman filter

evyatarluv, 2021-12-01:
Hi @TrivasZhang

There are some bugs with v3.3.3 on A1 When I'm trying to run the example_walk the robot doesn't move and the observations are {0}.
The same exmaple with v3.2 work fine.

TrivasZhang, 2021-12-01:
@evyatarluv 
Have you upgrade the A1_sport version to >=1.20?

TrivasZhang, 2021-12-01:
@talhen123 
Sorry for the raw data, can't be trans now.

evyatarluv, 2021-12-01:
> @evyatarluv Have you upgrade the A1_sport version to >=1.20?

I'm not sure about that.. how can I upgrade the A1_sport version?

TrivasZhang, 2021-12-01:
Ask our sales team for upgrade issue.

talhen123, 2021-12-01:
> @talhen123 Sorry for the raw data, can't be trans now.

That's fine, we're doing the project together

itaouil, 2021-12-01:
@TrivasZhang 

I saw that the 3.3.3 release was only targeted to the A1 robot and not the Aliengo.

Would it be possible to extend it also to the Aliengo robot, as that is the one that I have.

Sorry for not noticing that it was only for A1 from an early start.

TrivasZhang, 2021-12-01:
@itaouil 
Please wait for the next version of Aliengo sport mode. I will notice my colleague to keep up.

Lior-Falach, 2021-12-05:
@TrivasZhang 

Many thanks for the support.
Using the 3.3.3 SDK with the A_Sport 1.20 I can operate the low level example such as position and velocity and get the motorstate feedback, however, the highllevel example, e.g. example_walk  seems to be doing noting, no commands sent to the robot anf no feedback returns.
Would greatly appreciate any help on the subject.

Lior

TrivasZhang, 2021-12-06:
File example_walk, line16
`udp(8090, "192.168.123.11", 8082 ...)` needs change to `udp(8090, "192.168.123.161", 8082 ...)`
I should update this release to prevent confusion.

Lior-Falach, 2021-12-06:
@TrivasZhang 
the current command is udp(8090, "192.168.123.161", 8082 ...), anyway, I tried both i.p. and non of them worked...

littleggghost, 2021-12-06:
@Lior-Falach 
Have you rebuild the project after modify the codes?

Lior-Falach, 2021-12-06:
@littleggghost 
Sure, rebuild and added (uncommented ) a print motorstate command.

TrivasZhang, 2021-12-07:
I retest it without error. Did you forget to reboot the robot after upgrade?
Or, maybe try this [https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/3.3.4](url)

Lior-Falach, 2021-12-08:
@littleggghost @TrivasZhang 
Tried both 3.3.3 and 3.3.4 SDK version, all work in Low State yet both do not work in High State.
I returned to old version 3.2 and High State examples work.
A_Sport version is 1.20
Would greatly appreciate your support.

Lior

TrivasZhang, 2021-12-08:
I doubt if you have upgraded the A1_sport correctly. Could you paste a screenshot of the `/home/unitree` folder?

itaouil, 2021-12-25:
> @itaouil Caused by the sdk needs sport mode support the interface, please wait for the next version of Aliengo sport mode. I will notice my colleague to keep up.

Hi @TrivasZhang,

Any update on when will the AlienGo sport mode be available so as to obtain low sensor readings while in high level mode also for AlienGo?

closed 2022-06-24 by TrivasZhang

TextZip, 2023-10-20:
Hi @TrivasZhang , I do not think AlienGo still supports the reading of Motor States during High-Level Command. Can this please be added?

heartInsert, 2024-07-04:
guys ,  do  you  know  how to get  ` /gazebo/model_states `  and  `/aliengo_gazebo/joint_states`    in  a  real  robot ?  I   have  run   successfully in  Gazebo , but  I  am  stuck   in    how  to   run    subscribe  or  publish  these  in  a  real  robot .