unitreerobotics/unitree_legged_sdk #20: Running examples without sudo

antonilo, 2022-02-03:
Hi, 
Is it somehow possible to do memory locking without sudo? Basically, I would like to run the examples without requiring sudo,

Thanks a lot in advance!
Antonio

TrivasZhang, 2022-02-07:
adding the following lines to `/etc/security/limits.confg` might allow you to run the SDK without `sudo`.

```
<username> soft memlock unlimited
<username> hard memlock unlimited
<username> soft nice eip
<username> hard nice eip
```
Thanks @yxyang
Reference link https://github.com/yxyang/locomotion_simulation

antonilo, 2022-02-08:
thanks! For me it was also necessary to add the following two lines. 

```
username soft rtprio 99
username hard rtprio 99 
```

closed 2022-02-08 by antonilo