unitreerobotics/unitree_legged_sdk #23: Wrong values returned when receiving high state

itaouil, 2022-02-14:
Hi,

I tried the SDK version 3.3 and 3.3.1 in order to command with high level commands and received the respective high state the AlienGo robot, but with both SDKs, and all the examples I tried (example_walk.cpp, lcm_server_high.cpp) the logged high state attributes are completely wrong.

In the example below I printed the velocities (while the robot was standing and therefore not moving) and the velocities returned are not correct:

[image]

Any idea why this would be?

TrivasZhang, 2022-02-15:
That's because the sportMode on Aliengo is fall behind SDK-v3.3. We will fix it next few days.

TrivasZhang, 2022-02-15:
That will  catch up with SDK-v3.4.2

itaouil, 2022-02-16:
Ok, thanks!

I will keep an eye for the release in the next few days.

itaouil, 2022-02-27:
Any update on when the new SDK is going to be released?

This month or most likely next month? So that I can plan the next steps for the project I am working on.

Thanks :)

TrivasZhang, 2022-03-01:
Sorry for my late response.
I promise before this week Saturday.

itaouil, 2022-03-04:
Ok thanks.

jorgef1299, 2022-03-07:
Hi @TrivasZhang @itaouil!

I'm now starting to work with the AlienGO robot, and I do not understand which is the best SDK release to use. Should I use v3.3.1 as suggested in the README, or should I go for v3.4.2 instead?

Also, do I have to update the robot's firmware? Sport mode? How can I do that?

Thanks.

itaouil, 2022-03-07:
Hi @jorgef1299,

I guess it depends on whether your AlienGo robot was delivered with the updated sport mode firmware or not (a way to check is to see if you can activate sport mode and the respective stair mode climbing locomotions -> https://www.roscomponents.com/en/index.php?controller=attachment&id_attachment=506)

If you can activate the above locomotions, which means you have your firmware updated already, then you will have to wait for a new release of 3.4.2 which @TrivasZhang  is working on, but I am not sure when it will be released.

If you can't activate them, which means you don't have a firmware update and that your robot cannot tackle uneven terrains, you can use SDK 3.2 I think (this is the one I was using before the firmware update).

In order to update the firmware you have to contact the sales team and ask, but @TrivasZhang can correct me if I am wrong.

I hope this helps!

TrivasZhang, 2022-03-08:
Sorry for my late reply.
You can now ask our support team for Aliengo-sportMode-v1.0.18 and related sdk. This version has a shorter motorState than SDK v3.4.2 in HIGHLEVEL control.
Though we plan to support SDK v3.4.2, but met some old version issue. So please use this version for your need now.

TrivasZhang, 2022-03-08:
> 

Sure

TrivasZhang, 2022-03-08:
> So if I ask the support team for **Aliengo-sportMode-v1.0.18 and related sdk** I will be able to command the robot and receive high state information?

Sure

TrivasZhang, 2022-03-08:
Please notice me if meet problems.

itaouil, 2022-03-08:
> Please notice me if meet problems.

I will email the support team now and will keep you updated.

jorgef1299, 2022-03-10:
Thank you both for your replies.
So, if I use the **Aliengo-sportMode-v1.0.18** version will I be able to activate the new sport 3.0 mode and handle uneven terrains?

itaouil, 2022-03-10:
Yes.

You can contact their support team (https://www.unitree.com/booking) and they will send you the sport mode and respective SDK. But I think this is only in the case where you already have the robot with the firmware updated (i.e. you can tackle uneven terrains with it right now). If not, then they will have to send you some USB sticks to do the firmware update first (this was at least my case as the robot I have was not updated to turn on sport mode 3.0 back then)

jorgef1299, 2022-03-10:
Ok. I thought that using this new firmware would also add sport mode 3.0. 

I will contact the support line for more information...
Thank you

itaouil, 2022-04-14:
OK, problem solved.

Thanks :).

closed 2022-04-14 by itaouil

itaouil, 2022-04-26:
Hi @TrivasZhang 

I noticed that you introduced the **MotorStateLight** in the high state , but the values are all 0 (all the other values are correct). Would it be possible to obtain the actual values of the motors?

TrivasZhang, 2022-04-27:
Did the Aliengo-sportMode-v1.0.18 runs?

itaouil, 2022-04-27:
Yes sport mode 1.0.18 runs.

TrivasZhang, 2022-04-29:
But our support team test today and motor data seems fine. Would you contact support@unitree.cc for more details?

itaouil, 2022-05-01:
Thanks for testing it out again.

It turns out it was my mistake. Now I also receive the light motor state values.