unitreerobotics/unitree_legged_sdk #24: Go1 Pro, no reaction to command (unitree_legged_sdk, High-level)

devemin, 2022-03-10:
Hello unitree,

I got my Go1 Pro . It's very nice :-)

I tried to use this unitree_legged_sdk, and execute the 'example_walk' source.

I receive the state value (IMU or rotation or velocity).
But when I send the command (LED, torque, High-level mode), No reaction happen at LED or Motors...

[ Question ]
Is this the limitation of the model's firmware ?

Can I move the robot with C++ source ?

Could you tell me about this ?
Thanks.

https://twitter.com/devemin/status/1499662771520040966?s=20&t=cdRfjKntV7-kAMw_Zadlaw
devemin

devemin, 2022-03-10:
This is the source.

<details>
<summary>tried source. click this triangle.</summary>

```
/*****************************************************************
 Copyright (c) 2020, Unitree Robotics.Co.Ltd. All rights reserved.
******************************************************************/

#include "unitree_legged_sdk/unitree_legged_sdk.h"
#include <math.h>
#include <iostream>
#include <unistd.h>
#include <string.h>

using namespace UNITREE_LEGGED_SDK;

class Custom
{
public:
    // Custom(uint8_t level): safe(LeggedType::A1), udp(level){
    Custom(uint8_t level): safe(LeggedType::Go1), udp(8090, "192.168.123.161", 8082, sizeof(HighCmd), sizeof(HighState)){

    //192.168.123.161 (jibun) :port 8082 -> high mode de imu nado shutoku ok
    //192.168.123.10  (jibun) :port 8007 -> daturyoku sitesimau 
    //ichido daturyoku suruto, hukkatusinai ?

    // Custom(uint8_t level): safe(LeggedType::A1), udp(8090, "192.168.123.161", 8082, sizeof(HighCmd), sizeof(HighState)){
        udp.InitCmdData(cmd);
        udp.SetDisconnectTime(dt, 1);
        // udp.SetDisconnectTime(0, 0);
    }
    void UDPRecv();
    void UDPSend();
    void RobotControl();

    Safety safe;
    UDP udp;
    HighCmd cmd = {0};
    HighState state = {0};
    int motiontime = 0;
    float dt = 0.002;     // 0.001~0.01
};

void Custom::UDPRecv()
{
    udp.Recv();
}

void Custom::UDPSend()
{  
    udp.Send();
}

void Custom::RobotControl() 
{
    motiontime += 2;
    udp.GetRecv(state);
    // printf("%d   %f\n", motiontime, state.forwardSpeed);
    // printf("%f %f \n", state.imu.quaternion[0], state.imu.quaternion[1]);
    // printf("%f %f \n", state.imu.rpy[1], state.imu.rpy[2]);
    printf("%f %f %f %f %f\n", state.imu.rpy[1], state.imu.rpy[2], state.position[0], state.position[1], state.velocity[0]);

    cmd.mode = 0;      // 0:idle, default stand      1:forced stand     2:walk continuously
    cmd.gaitType = 0;
    cmd.speedLevel = 0;
    cmd.footRaiseHeight = 0;
    cmd.bodyHeight = 0;
    cmd.euler[0]  = 0;
[... 143 lines cut ...]
    return 0; 
}

```

</details>

```

    cmd.led[0].r = 255;
    cmd.led[0].g = 0;
    cmd.led[0].b = 0;
```

```
        cmd.mode = 2;
        cmd.gaitType = 2;
        cmd.velocity[0] = 0.4f; // -1  ~ +1
        cmd.yawSpeed = 2;
        cmd.footRaiseHeight = 0.1;
        // printf("walk\n");

```

LED and Motor Didn't move .

suiduck, 2022-03-30:
We have the same issue as @devemin. We tried all examples but none of them work as expected, although UDP connection seems work fine and is able to receive the values from IMU. We tried to run the examples in the internal Raspberry and from an external device through an Ethernet cable, but both are not working. Is there something tricky we missed? Please help. Thanks. 

我們是大學的研究部門，同樣遇到問題。程式能收到UDP發出來的IMU數值，但是機器狗對發出的CMD完全沒有反應。我們嘗試用內建的樹莓派及網線接駁另一台電腦執行示範程序，兩者同樣失敗。請問我們是否遺留了什麼？請幫忙，謝謝。

xinsongyan, 2022-03-30:
We got the same issue here... tried sdk3.4.2, sdk3.5.1 and the sdk on raspberry pi, none of them can send commands to the robot.

xinsongyan, 2022-03-30:
> Please make sure that the Go1 is an EDU version.

Yeah, I think we do have an edu version.

suiduck, 2022-03-30:
> Please make sure that the Go1 is an EDU version.

It should be the Edu version as our University bought the Go1 for a while and we have the development docs. And as I remembered Unitree team members Ackles and Irving (sorry if I misspell their names) told us we can ask in here if we got development problems. Thanks.

TrivasZhang, 2022-03-30:
OK, let check it one by one.
1. Is it EDU version?
2. After reboot, can this robot auto standup  and control by gamepad? This will diagnose if the sport mode still works.
3. $ cd ~/Unitree/autostart/sportMode
    $./bin/Legged_sport -v
    Then check the sport mode version. It should seems like _Legged_sport version:x.xx_
    I need this version to assure the according firmware version.
4. Can the origin sdk examples work? You can find it at ~/Unitree/sdk/unitree_legged_sdk
5. When tried the SDK, please quit sport mode to normal node, then run _example_velocity_ to check if low-level control work.

suiduck, 2022-03-31:
> OK, let check it one by one.
>
> 1. Is it EDU version?
> 2. After reboot, can this robot auto standup  and control by gamepad? This will diagnose if the sport mode still works.
> 3. $ cd [...]

Thanks for your prompt reply. We will go through the process you suggested. 

Just a quick question, our professor just reminded me he downloaded the firmware and packages from Unitree website and updated the robot last month. Knowing that we got some exclusive materials for the edu version, would the firmware and packages downloaded from the public website (it doesn’t specify whether the updates are for air, pro, or edu) mess things up? Thank you.

TrivasZhang, 2022-03-31:
The update about firmware and software are not distinguish AIR or EDU version. The version message is flash loaded when manufactured, and upper software has different reaction to different version.
Did the low-level works before professor upgrade the firmware?

TrivasZhang, 2022-03-31:
Use methods at [Go1_update]( https://www.unitree.com/go1_update) to check versions. A screenshot is good.

Affonso-Gui, 2022-04-05:
I was also having troubles with this. However, the robot seemed to work fine when setting the levelFlag to 0.
```cpp
    cmd.levelFlag = 0;
```

It also seems to work with `cmd.levelFlag = LOWLEVEL;` or `cmd.levelFlag = TRIGGERLEVEL;` or in fact with anything other than the default `HIGHLEVEL` value.

@devemin

devemin, 2022-04-05:
Hello @Affonso-Gui 

I got success to moving with example_walk.cpp (High-level) !!
Grateful thanks!

But example_position / velocity / torque was not worked.

Can you give me some advice ? 
Thanks.

[supplementary explanation]

I added 'cmd.levelFlag  = 0; (or LOWLEVEL or TRIGERLEVEL )' at Line No.64 in example_position.cpp.
(and I tried 'cmd.motorCmd[FR_0].mode = 0x0A; ( or 0x00) ' )

But It is not work.

https://github.com/unitreerobotics/unitree_legged_sdk/blob/376b4f7d874ac9a42bed9099d2dab217a79eeb0e/examples/example_position.cpp#L64

Affonso-Gui, 2022-04-05:
Actually I haven't tried the other examples yet, but I'll try to have a look!

TrivasZhang, 2022-04-07:
When command with low-level control, please make sure the robot has quit sport mode, 
and command with high-level control, please make sure the robot is under sport mode.

devemin, 2022-04-08:
@TrivasZhang 

Hello.

In the Go1 documents, No description about how to change the mode (sport / normal).
It confuse me.

Could you tell me about how to change the mode ?

(And maybe Unitree should modify or add the document :-) )

Masahito

Iron-Fatty, 2022-04-08:
1. switch sport mode（HighLevel mode） to normal mode（LowLevel mode）
L2+B (switch to damping mode)
L1+L2+START (switch to normal mode)
L2+B twice (swtich to damping mode with noise）
L2+A twice (switch to stand postion)

2. switch normal mode（LowLevel mode） to sport mode（HighLevel mode）
L2+B (switch to damping mode)
L1+START (switch to sport mode)

devemin, 2022-04-08:
@Iron-Fatty 
I got it, thanks.

I tried this. 
When I pushed the button L1+L2+START, the robot was quiet.

And I executed 'sudo ./example_position ', But It was not work.

I think this is a function limitation resulting from differences in product lineup models maybe.
(I am using Go1 pro. Maybe low-level mode need Edu ).

I will use High-level mode. 
Thank you!

closed 2022-04-21 by littleggghost

MAVProxyUser, 2022-12-18:
For anyone else trying to follow along. I have added some more detail here: 
https://github.com/MAVProxyUser/YushuTechUnitreeGo1/blob/main/SDK.md

You need to pay attention to versions, IP addresses, and that the "A1" driver has not been selected. I can't figure out why Unitree staff does not make better documentation for using High Level control of the dogs on non-EDU versions. 

"no you can not do that" seems to be what they tell us instead, why!?