unitreerobotics/unitree_legged_sdk #30: Bus error on A1's Raspberry Pi with 3.3.2 release

Cromvell, 2022-04-07:
Hello. I've encountered issue with version of the code that interfacing with SDK through pybind11. When attempting to access state received from the robot, specifically arrays like state.imu.quaternion, we end up getting null value for the array pointer and Bus error as the result. Interestingly, it happens on version 3.3.2 with C++ examples as well. Seems like change to std::array in comm.h affecting the whole API. Is it possible to fix that, along with keeping python interface? Thanks.

TrivasZhang, 2022-04-07:
Did this error also happens on A1's x86? It locates on the rear of raspi.

TrivasZhang, 2022-04-07:
Another question is, could you compare the comm.h file between v3.3.2 and the version which could run before.

Cromvell, 2022-04-08:
On x86 everything works fine.

The difference between 3.3.2 and other SDK version that works on ARM is precisely the use of std::array instead of static arrays in comm.h. Here's the diff: https://www.diffchecker.com/uNINjE1x

I've checked that after this change in working SDK, the error starts to appearing.

TrivasZhang, 2022-04-22:
Hi @Cromvell 
The member order of the class HighCmd and HighState is different between v3.3.2 and the version which could run before.