unitreerobotics/unitree_legged_sdk #33: GO1 PRO

JonathanMortes, 2022-04-29:
Hi, is it possible to be used on GO1 PRO?

TrivasZhang, 2022-05-05:
I'm sorry that this sdk is only for edu version now.

closed 2022-05-07 by JonathanMortes

MAVProxyUser, 2022-12-18:
@TrivasZhang why does Unitree continue to say this over and over and over? Is it for more money for the sales team? 

By now you all should correct documentation to show that ALL Go1 robots will work with High Level control. 

Stop making sales team wallets fatter. 

@JonathanMortes YES the SDK *IS* possible on Pro, in limited fashion, as has been circulated by other people tinkering on the internet. I can not figure out why Unitree keeps reciting this myth that you can not "program" the Go1 below EDU model. It simply is not true. You just need the right version, and the proper parameters, when compiling. Also you must understand that there is *limited* support of the SDK, but you can 100% control the dog still, and build ROS logic around this. 

https://github.com/MAVProxyUser/YushuTechUnitreeGo1/blob/main/SDK.md