unitreerobotics/unitree_legged_sdk #34: How to get the motor state from High Level mode in Go1 Edu

HBSIM0, 2022-05-04:
Hello. Unitree

I am using Go1 Edu, and I want to get motorState and footForce values of HighState from sdk v3.5.1.
However, these values can only be obtained from the Low State.
Is there a way to get the MotorState and footForce values from the high level mode?

TrivasZhang, 2022-05-05:
Do you mean that the footForce value is zero  in high-level or can't find this variety in high-level?
And do you know the sport mode version? A screenshot with APP will tell you that.

HBSIM0, 2022-05-06:
> Do you mean that the footForce value is zero in high-level or can't find this variety in high-level? And do you know the sport mode version? A screenshot with APP will tell you that.

What I mean that is the footForce value is zero in high-level.
And sport mode is 1.37.0
[image]

TrivasZhang, 2022-05-06:
1. Does the rubber tube connect with the foot? 
2. Could you find a typeC wire? Connect the forward typeC port (on robot) with your computer, use a serial assistant software to check the prints. If the footforce sensor works, then the corresponding values will change when feels pressure.

HBSIM0, 2022-05-06:
> 1. Does the rubber tube connect with the foot?
> 
> 2. Could you find a typeC wire? Connect the forward typeC port (on robot) with your computer, use a serial assistant software to check the prints. If the footforce sensor works, then the corresponding values will change when feels pressure.

1. The rubber tube is connected to the foot.
2. I checked that the foot force value is printed on the serial port log.

The point I'm talking about is that the foot force value is only output when in low mode.
I use different udp functions as shown in the example  below.

low mode(example_postion)
UDP(uint8_t level, HighLevelType highControl = HighLevelType::Basic); 

high mode(example_walk)
UDP(uint16_t localPort, const char* targetIP, uint16_t targetPort, 
            int sendLength, int recvLength, bool initiativeDisconnect = false, RecvEnum recvType = RecvEnum::nonBlock);

TrivasZhang, 2022-05-07:
I checked and find this is an error about variety value assignment.
I will fix it in next sport mode version.
Really thanks!

TrivasZhang, 2022-05-17:
Sport mode v1.38 is avaliable now, you can download it at
https://www.unitree.com/go1_update

HBSIM0, 2022-05-19:
> Sport mode v1.38 is avaliable now, you can download it at https://www.unitree.com/go1_update

I updated the software and checked the motor state in high-mode.
Thank you.
I closed the issue.

closed 2022-05-19 by HBSIM0