unitreerobotics/unitree_legged_sdk #35: In high-level mode , cmd.mode 2 does not work when set cmd.mode 0

HBSIM0, 2022-05-24:
Hello Unitree

I am implementing a moving robot using example_walk.cpp.
The code  below does not work.

//example_walk.cpp 121 line
if(motiontime > 20000 && motiontime < 24000){
    cmd.mode = 2;
    cmd.gaitType = 1;
    cmd.velocity[0] = 0.2f; // -1  ~ +1
    cmd.bodyHeight = 0.1;
    // printf("walk\n");
}

It works fine until line 120.

robot :  Go1 Edu
sdk version : v3.5.1
Sport mode : 1.37.0
hardware : 0.1.9
firmware : 0.1.37

jjq556, 2022-05-26:
Hi, sorry for the inconvenience brought to for. You need to change the"0.1" in the  cmd.velocity[0] = 0.1f; // -1 ~ +1 into a bit larger number such as 0.2. The explaination of  -1 ~ +1 is a wrong version,pls ignore it. If you still have some questions,pls let me know.

HBSIM0, 2022-05-26:
It's very simple
Thanks

closed 2022-05-26 by HBSIM0