unitreerobotics/unitree_legged_sdk #37: Receive HighState feedback at 1kHz

qw8z5fUB, 2022-05-31:
Hi there, I'd like to log a feature request.

I note that it is now possible to send high level commands and receive low level feedback (motor state). This is useful for implementing state-estimation-related systems while still utilising the high level command interface.

However I've noticed that when using the high level interface, the state feedback is only sent at 50 Hz, while in low level mode it is sent at 1kHz. This significantly limits the accuracy of state estimation that is achievable in high level mode compared to low level mode.

I'd request that state feedback be sent at 1kHz even when operating in high level mode, such that we can achieve the same state estimation accuracy in high level mode that we can achieve in low level mode. 

My setup is as follows:
```
Robot go1 edu HW 1.0.9
Software 0.1.35
Legged_sport version:1.36
unitree_legged_sdk version a5df2e4471b3c34 (with small local modifications).
```

TrivasZhang, 2022-06-15:
OK, next version.

Wsser, 2022-11-28:
emmmm

xyq1223, 2023-03-14:
Hello,our new version is under development, please wait