unitreerobotics/unitree_legged_sdk #39: Sometimes the Go1 app BMS information is not visible.

HBSIM0, 2022-06-02:
Sometimes Go1 App BMS information is not visible.
At this time, I checked the SDK's GetRecv (HighState) and found that all the information was zero.

[image]

[image]

TrivasZhang, 2022-06-15:
Did this bms information works at some times, or it always nothing?

HBSIM0, 2022-06-16:
It happens sometimes

TrivasZhang, 2022-06-16:
If sport mode process (Legged_sport) is killed or quit, then the information is unavaliable, until this process rerun again.

HBSIM0, 2022-06-17:
How do I check if the sport mode process is alive?

If BMS is not available, BMS information is not available even if the robot reboots and waits for more than 3 minutes. 
When the battery is fully charged, BMS information is available.

TrivasZhang, 2022-06-17:
Next time if the BMS info missing, then you can do the low-level control to print the bms state.
This will locate the problem.

HBSIM0, 2022-06-17:
I don't think I can solve the zero on the app.

TrivasZhang, 2022-06-17:
I need find out this problem happens at high mode or low mode, so when informations gone next time, exit sport mode and use low mode to print. If low mode works then it's high mode problem.

MAVProxyUser, 2023-01-03:
This is the same issue I mailed support about a month or so ago. I was told it was a known bug. I am on a MAX model, which I understand is basically a pro with a 4g modem, so I am unable to move to low-level as suggested above. My BMS output has been steady broken for months on end.

maggusscheppi, 2023-01-16:
This issue seems to have a root cause in the electronic on the battery. You can read out values from battery via usb-c with 921kbit on the charging station. With a defect battery you receive no values.

closed 2023-03-31 by HBSIM0