unitreerobotics/unitree_legged_sdk #40: installation error

JetBelson, 2022-06-07:
PC OS: Ubuntu20.04.

After following the instructions from the manual, all dependencies were installed successfully, and the ping operation works. But the SDK installation has the below error.

Any suggestion?

_/usr/bin/ld: skipping incompatible /home/XXX/software/unitree_legged_sdk/lib/libunitree_legged_sdk_arm64.so when searching for -lunitree_legged_sdk_arm64
/usr/bin/ld: cannot find -lunitree_legged_sdk_arm64
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/lcm_server.dir/build.make:84: lcm_server] Error 1
make[1]: *** [CMakeFiles/Makefile2:90: CMakeFiles/lcm_server.dir/all] Error 2
make: *** [Makefile:84: all] Error 2_

JetBelson, 2022-06-07:
SDK Version 3.5.1

HKU-CICID, 2022-06-07:
the error is due to the link library, need to match it with your machine by choosing libunitree_legged_sdk_arm64 or libunitree_legged_sdk_amd64.

closed 2022-06-07 by JetBelson

pab47, 2022-06-20:
Try changing in CMakelists.txt this line: set(EXTRA_LIBS -pthread libunitree_legged_sdk_arm64.so lcm) to 
set(EXTRA_LIBS -pthread libunitree_legged_sdk_amd64.so lcm)

kenloumixx, 2022-08-02:
Hello,

Have you solved this problem? 

I am suffering from the same issue :-(

Chen001117, 2022-08-12:
> Try changing in CMakelists.txt this line: set(EXTRA_LIBS -pthread libunitree_legged_sdk_arm64.so lcm) to set(EXTRA_LIBS -pthread libunitree_legged_sdk_amd64.so lcm)

This change fix my problem. Thanks!