unitreerobotics/unitree_legged_sdk #41: Feed-forward torque control mode?

egordv, 2022-06-15:
Hello! 
I want to use the feed-forward torque control method on Go1 robot with unitree_legged_sdk v3.4.1
The control law that I need is: `tau = tau_des + P * (q_des - q) + D * (v_des - v)`

As far as I understood from the examples, I can do the classic torque control (`tau = tau_des`) with:
```
cmd.motorCmd[k].q = PosStopF;
cmd.motorCmd[k].dq = VelStopF;  
cmd.motorCmd[k].Kp = 0;  
cmd.motorCmd[k].Kd = 0;  
cmd.motorCmd[k].tau = tau_des;  
```
Also I can do the classic position control (`tau = P * (q_des - q) + D * (v_des - v)`)  with:
```
cmd.motorCmd[k].q = q_des;
cmd.motorCmd[k].dq = v_des;
cmd.motorCmd[k].Kp = Kp;
cmd.motorCmd[k].Kd = Kd;
cmd.motorCmd[k].tau = 0;
```
But is there any specific structure of motorCmd designed to perform a feed-forward torque control mode with the control law `tau = tau_des + P * (q_des - q) + D * (v_des - v)`?

TrivasZhang, 2022-06-15:
Yes, you are right.
A notable point is, the control bandwidth is different.
The motor driver runs on MCU and has high control bandwidth, about 10~20kHz, so when [q dq Kp Kd tau] is set to motor, then the final torque responds rapidly.
But when you use a feedback value to generate torque, as you said 'classic position control'
```
cmd.motorCmd[k].q = 0;
cmd.motorCmd[k].dq = 0;
cmd.motorCmd[k].Kp = 0;
cmd.motorCmd[k].Kd = 0;
cmd.motorCmd[k].tau = P * (q_des - q) + D * (v_des - v);
```
The control bandwidth will down to about 1khz cause of communication speed.

TrivasZhang, 2022-06-15:
Sure you can. It will be 10kHz.

pab47, 2022-06-20:
I think it would be a mix of the two: the PD control will be at 10 KHz assuming (Kp, Kd, q, dq are not being updated) but the tau_des will be at 1 Khz (assuming that is is updated in the main loop).

TrivasZhang, 2022-06-21:
> I think it would be a mix of the two: the PD control will be at 10 KHz assuming (Kp, Kd, q, dq are not being updated) but the tau_des will be at 1 Khz (assuming that is is updated in the main loop).

Yes, if tau is a variety then tau is 1khz, if tau is constant then is 10khz.

TrivasZhang, 2022-06-24:
Feel free to reopen it.

closed 2022-06-24 by TrivasZhang

egordv, 2022-06-30:
Hello!
I've checked this approach on real Go1 and everything works as expected. 
Control bandwidth for q/dq commands is 10kHz, and for tau is 1kHz, which really helps to solve stability issues when working with high control gains.
Thank @TrivasZhang @pab47 for your support!