unitreerobotics/unitree_legged_sdk #42: empty state variable

smileyenot983, 2022-06-16:
We are currently running on of the examples (example_walk.cpp) via a wired connection, robot does all the actions, defined in the code, but the state variable is filled with zeroes all the time:

By state i mean HighState

HighState state = {0};
udp.GetRecv(state);

smileyenot983, 2022-06-16:
robot firmware 1.19

sdk version 3.3.2

TrivasZhang, 2022-06-16:
Can i see the app screecshot with versions?

smileyenot983, 2022-06-16:
> Can i see the app screecshot with versions?

What do you mean by app screenshot? I have unitree_legged_sdk, which is compiled and i am running example_walk. My PC is connected to the robot via ethernet cable. 

here is the code part, where i am printing the state(printing was just commented and i have uncommented it):
void Custom::RobotControl() 
{
    motiontime += 2;
    udp.GetRecv(state);

    printf("%f %f %f %f %f\n", state.imu.rpy[1], state.imu.rpy[2], state.position[0], state.position[1], state.velocity[0]);

    cmd.mode = 0;
    cmd.gaitType = 0;
    cmd.speedLevel = 0;
    cmd.footRaiseHeight = 0;
    cmd.bodyHeight = 0;
    cmd.euler[0]  = 0;
    cmd.euler[1] = 0;
    cmd.euler[2] = 0;
    cmd.velocity[0] = 0.0f;
    cmd.velocity[1] = 0.0f;
    cmd.yawSpeed = 0.0f;

    if(motiontime > 0 && motiontime < 1000){
        cmd.mode = 1;
        cmd.euler[0] = -0.3;
        // cmd.velocity[0] = 0.2f;
    }

TrivasZhang, 2022-06-16:
I want the app help me to ensure these verisons, like:
https://user-images.githubusercontent.com/97996197/171523833-d9bc08f9-0562-430e-bbae-eb5bafc4680e.jpg

smileyenot983, 2022-06-16:
> I want the app help me to ensure these verisons, like: https://user-images.githubusercontent.com/97996197/171523833-d9bc08f9-0562-430e-bbae-eb5bafc4680e.jpg

I forgot to mention that we have unitree a1, we do not have such application as in your screenshot

TrivasZhang, 2022-06-17:
My colleague will try this next week.

fLattes, 2022-06-20:
I have the same problem with the Go1 EDU, the state values are all 0 but the robot moves. I'm using the WiFi connection.

TrivasZhang, 2022-06-20:
If under AP connection, then the IP should change to "192.168.12.1".

fLattes, 2022-06-20:
Thank you, it works.
And using ROS, what can I change to have the IMU data?

fLattes, 2022-06-21:
When I launch the program "state_sub" , after launching "roslaunch unitree_legged_real real.launch ctrl_level:=highlevel" and for example "rosrun unitree_legged_real example_walk", all the data are equal to 0, but the robot moves.

TrivasZhang, 2022-06-21:
Then try to modify this line
https://github.com/unitreerobotics/unitree_ros_to_real/blob/master/unitree_legged_real/src/exe/ros_udp.cpp#L28

TrivasZhang, 2022-06-24:
Feel free to reopen it.

closed 2022-06-24 by TrivasZhang