unitreerobotics/unitree_legged_sdk #46: for Python 3.8 lib

devemin, 2022-06-27:
Dear Unitree

I tried the python commit, It works! (Python 3.7)

And I hope Could you add the ARM lib (.so)  for Python 3.8 ?
I need to use it with another host (Python3.8 / Ubuntu 20.04 / Raspi 4).

https://github.com/unitreerobotics/unitree_legged_sdk/tree/master/lib/python/arm64

Thanks :-)

TrivasZhang, 2022-06-28:
Did [this lib](https://github.com/unitreerobotics/unitree_legged_sdk/tree/master/lib/python/arm64) unsatisfy your need?
Because this lib could work under raspbian buster, 64bit.

devemin, 2022-06-28:
@TrivasZhang 
Thanks for you reply.

Maybe Is the lib file for python 3.7 ?

I think it can't use with python 3.8, right ?
I try it with raspi4 (ubuntu20.04/64bit/python 3.8 ), but It appeared the import error.

>>cd ./build
>>cmake ..
>>make -j4
>>//change the source, "sys.path.append('../lib/python/arm64')"
>>
>>export LD_LIBRARY_PATH=/path-to-unitree_legged_sdk/lib/cpp/arm64:$LD_LIBRARY_PATH
>>python3 ../example_py/example_walk.py
>>
>>ModuleNotFoundError: No module named 'robot_interface'

I need to use with python 3.8.

or, Should I install python 3.7 ?

devemin, 2022-06-28:
fix

>>export LD_LIBRARY_PATH=/path-to-unitree_legged_sdk/lib/cpp/arm64:$LD_LIBRARY_PATH

>>export LD_LIBRARY_PATH=../lib/cpp/arm64:$LD_LIBRARY_PATH
[image]

TrivasZhang, 2022-06-28:
So could this 3.7 version lib works with python3.8?
I know the 3.8 is more versatile, but the known issue is that many delivered robots are python 3.7.

devemin, 2022-06-28:
No, It fix only the path string.
I tried, but It didn't work with python 3.8 yet. (It showed as screenshot.)

If unitree can provide the lib for python 3.8, It is convenient.
But If can't, I will try to add python 3.7.

Thank you, @TrivasZhang

TrivasZhang, 2022-06-28:
I'll add 3.8.

TrivasZhang, 2022-06-28:
> fix
> 
> > > export LD_LIBRARY_PATH=/path-to-unitree_legged_sdk/lib/cpp/arm64:$LD_LIBRARY_PATH
> 
> > > export LD_LIBRARY_PATH=../lib/cpp/arm64:$LD_LIBRARY_PATH
> 
> [image]

Did absolute path not work?

devemin, 2022-06-28:
I did again with absolute path, but same result.

( Off course Go1's raspi can work with this.)

[image]

TrivasZhang, 2022-06-28:
`cd` to example_py folder then `python3 example_walk.py`

devemin, 2022-06-28:
Thanks @littleggghost 
[image]

same

I think because of module version.

TrivasZhang, 2022-06-28:
Use `sys.path.append('../lib/python/arm64')`
By the way, where did you find this raspi4 (ubuntu20.04/64bit/python 3.8 ) system image?

devemin, 2022-06-28:
Next, I try to rename the lib.

```
mv ./robot_interface.cpython-37m-aarch64-linux-gnu.so robot_interface.so
```

Then version mismatch log appeared.

>>> import robot_interface
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: Python version mismatch: module was compiled for Python 3.7, but the interpreter version is incompatible: 3.8.10 (default, Mar 15 2022, 12:22:08) 
[GCC 9.4.0].

[image]

Thanks.

TrivasZhang, 2022-06-29:
Please wait for the 3.8 version lib.

devemin, 2022-06-29:
Dear TrivasZhang,

Thanks for helping me. I really appreciate it.
I will wait for it.

devemin

TrivasZhang, 2022-06-30:
Good to hear that! If everything stable, I'll make a release.

devemin, 2022-06-30:
I will use and test this for a week :-)

closed 2022-07-13 by devemin

Dorianproject, 2023-03-13:
Is there a python 3.10.6 solution for the: " No module named 'robot_interface' " problem 
?

xyq1223, 2023-03-14:
Hello,we have provided the python_wrapper,please check and any version of Python （>3.0） can be used to control the robot dog
(https://github.com/unitreerobotics/unitree_legged_sdk/tree/go1/python_wrapper)

ketulpanchal, 2023-12-28:
> Hello,we have provided the python_wrapper,please check and any version of Python （>3.0） can be used to control the robot dog (https://github.com/unitreerobotics/unitree_legged_sdk/tree/go1/python_wrapper)

Hi, I am using python3.10.12 and I am getting the same error saying "no module named robot_interface" Can I please know how to use python_wrapper to solve this error? Or do I need to use different version of Python