unitreerobotics/unitree_legged_sdk #52: A1: Changing Robot State from Normal Mode to Sport Mode through SDK

sieversb, 2022-08-05:
I am using the SDK vesion 3.3.1 and was wondering if it is possible to use the SDK to change the roboter state from normal mode into sports mode. At the moment, we use the default remote control, which is delivered with the robot, to switch from normal mode to sport mode after we power up the robot. However, we want to develop in sports mode and using the remote control to first switching the robot into sports mode after every boot is a bit unsatisfying in the long-term. Is there another way to we might miss?

JonasFovea, 2022-12-06:
Hi!

I'm only familiar with the GO1 robot, but I'd guess they use a similar system.
On the Go1, there is the Raspberry Pi and the Main Control Board, which work together.
The Sport Mode is handled by the Pi and the Joint control is handled by the MCB.

To activate Sport Mode, a shell script is run, which then repetitively restarts the sport mode program, in case it fails.

My guess is, that they've mapped this script to the remote control, to turn it on or off.

The SDK on the other hand connects either to the Pi or the MCB, depending on of you are using high- or low-level control. 
The high-level works only if the sport mode is running on the Pi and the low-level control only works, if the sport mode isn't running on the Pi since the ports would otherwise be blocked.

So the UDP messages provided by the SDK are im my eyes not suitable for toggling between normal and sport mode.
You could try to run another program on the Pi (should be automatically started), which receives messages on a different IP port.
And that program could then either start or kill the sport mode process.

xyq1223, 2023-03-13:
Sorry, our SDK does not support this function

RN0H, 2024-05-07:
Is there any solution to this? I am able to change the levels (low/high) through the SDK, however, I am in need of a way to even change the modes (normal/sport). It seems that running the A1_sport exe itself isn't the solution to initiate Sport mode, since it requires L1 + start to start it. Also for triggering it off, it requires L1+L2+start. Is this achievable?

JonasFovea, 2024-05-07:
I don't think that's possible.
The sport mode is an executable on the robot, which is started via a script in the Autostart directory.

You would need to kill that process and restart it.

So maybe write a program, which listens to such commands on a socket and then does the corresponding action.
And that program itself can be started when booting the robot.

RN0H, 2024-05-09:
Well killing/restarting the exe is not a problem. The problem is that the exe when initiated, waits for L1 + start from the RC, in order to start sport mode. The exe itself does not launch sports mode I believe.