unitreerobotics/unitree_legged_sdk #53: /unitree/state returns all zero values & LCM Time out.

kenloumixx, 2022-08-11:
Hello,

First of all, thank you for the robots and the software tools :)

I made a connection between unitree A1 robot and the laptop,

the rostopic lists appeared and ping test was successful.

However, when I ran 'rostopic echo /unitree/state/' that is supposed to show the joint positions and other things, all the parameter values returned zeros:
[image]

And at the same time, the error message that LCM connection has timed out appears, as below:
[image]

Is there anyone who faced this problem and solved it?

Thanks in advance!

closed 2022-08-12 by kenloumixx

graiola, 2022-10-17:
Hello, how did you solve this issue? Thank you

Charlescai123, 2023-11-09:
Hello, I'm also suffering from the issue and just wondering how you solved it, thanks!

giuseppecialdella, 2023-11-13:
Hi, I have the same problem using the examples of the unitree_legged_sdk. 
When the robot is standing up in High Level and I try to print the HighState values it returns all zeros. 
How can i solve?
Moreover, what (which node or launch file of which repository) do i need to launch to obtain the topic /unitree/state in ROS? Thanks in advance!