unitreerobotics/unitree_legged_sdk #54: Rear left calf motor does not receive command

anahrendra, 2022-08-19:
Hi! 

We are doing a research with the Unitree A1 robot. So we made our own locomotion controller and run it on an Intel NUC placed on top of the robot. We connect the NUC via LAN cable to the LAN port on the Jetson NX of the robot. 

Everything works fine for most conditions, including stairs and rough terrain. However, we have a strange issue. Everytime the robot climbs a relatively steep slope (maybe around 18 degree), the rear left calf suddenly crippled.

When we log the data of the motor, we noticed that for that leg only, the control mode became **0x00 (or 8)**, which according to the A1 Dev. guide indicates **'electronic braking mode'** due to disconnection protection. However, I couldn't find further details about it.

We checked the LowState and got this data at the time the leg crippled
`
Temperature, mode, reserve, torque, position, velocity
48.000000 10.000000 0.000000 24.382812 -1.543421 0.541724
48.000000 10.000000 0.000000 24.347656 -1.542578 0.028331
49.000000 10.000000 0.000000 24.347656 -1.540724 0.637019
48.000000 10.000000 0.000000 24.562500 -1.537394 1.049966
48.000000 10.000000 0.000000 17.593750 -1.530820 0.804430
48.000000 10.000000 0.000000 16.777344 -1.530778 -0.183723
48.000000 10.000000 0.000000 16.847656 -1.530736 0.254121
48.000000 10.000000 0.000000 17.308594 -1.528418 0.723729
48.000000 8.000000 4.000000 3.234375 -1.523024 1.309238
49.000000 8.000000 4.000000 5.367188 -1.531073 -2.379808
48.000000 8.000000 4.000000 3.339844 -1.539923 -1.243990
49.000000 8.000000 4.000000 1.847656 -1.544179 -0.746909
48.000000 8.000000 4.000000 1.101562 -1.547508 -0.522837
49.000000 8.000000 4.000000 0.531250 -1.549616 -0.291896
49.000000 8.000000 4.000000 -0.035156 -1.550248 0.097012
48.000000 8.000000 4.000000 0.000000 -1.549953 -0.117617
48.000000 8.000000 4.000000 -0.035156 -1.549742 0.084135
`

It looks like everything did not surpass the motor limits. So I have no idea why this problem happens.

Notes:
1. The NUC is powered using an external battery and regulator
2. Realsense camera is connected directly to the NUC

Thanks in advance for your help!

TrivasZhang, 2022-08-19:
As shown in mode and reserve, this motor is under temperature protection, and turn off it's own power.
Did this motor temperature high?

anahrendra, 2022-08-19:
Hi! Thanks for your quick reply.

As the log that I attached above, the temperature of the motor is 48. And when i touched the motor, i can feel that the other motor (front legs) feel even warmer compared to this motor.

So that means, it is certainly not about the disconnection problem?

TrivasZhang, 2022-08-20:
Maybe the temperature sensor fails. 
Please contact our sales for more help.