unitreerobotics/unitree_legged_sdk #55: Where is the modified MIT Cheetah source code for Legged_sport?

MAVProxyUser, 2022-08-22:
At the very least you must supply the MIT license with the dog upon purchase. 

Some of us would like to modify the things that you borrowed from MIT, please share the source code too! 

https://github.com/mit-biomimetics/Cheetah-Software/blob/master/LICENSE
https://github.com/mit-biomimetics/Cheetah-Software/issues/2

You can see the MIT Cheetah use in Legged_sport
```
$ grep cheetah Go1_2022_05_11_e0d0e617/ -ri
Binary file Go1_2022_05_11_e0d0e617//raspi/Unitree/autostart/sportMode/bin/Legged_sport matches
```

Please do not delete this like you did my 'MIT backflip' issue: https://github.com/unitreerobotics/unitree_legged_sdk/issues/47 

Cheetah is in Go1, how do we share his code? 
```
$ strings Go1_2022_05_11_e0d0e617//raspi/Unitree/autostart/sportMode/bin/Legged_sport | grep -i cheetah  | sort | uniq
22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeE
CHEETAH_3
CHEETAH_3_ZERO_OFFSET
Cheetah3.h
MINI_CHEETAH
MiniCheetah.h
PeriodicMemberFunction<MiniCheetahHardwareBridge>
[Backflip DataReader] Setup for mini cheetah
[Cheetah Test] Test initialization is done
[Front Jump DataReader] Setup for mini cheetah
[Offline Motion DataReader] Setup for mini cheetah
_Z13buildCheetah3IfE9QuadrupedIT_Ev
_Z16buildMiniCheetahIdE9QuadrupedIT_Ev
_Z16buildMiniCheetahIfE9QuadrupedIT_Ev
_ZL21CHEETAH_3_ZERO_OFFSET
_ZN13LegControllerIdE20setMaxTorqueCheetah3Ed
_ZN13LegControllerIfE20setMaxTorqueCheetah3Ef
_ZN20CheetahVisualizationC2Ev
_ZN20CheetahVisualizationC4Ev
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeE3runEv
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeE4initEv
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeE7cleanupEv
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeEC2EP19PeriodicTaskManagerfNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEiMS0_FvvEPS0_
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeEC4EOS1_
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeEC4EP19PeriodicTaskManagerfNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEMS0_FvvEPS0_
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeEC4EP19PeriodicTaskManagerfNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEiMS0_FvvEPS0_
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeEC4ERKS1_
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeED0Ev
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeED1Ev
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeED2Ev
_ZN22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeED4Ev
_ZN23StateEstimatorContainerIdE3runEP20CheetahVisualization
_ZN23StateEstimatorContainerIfE3runEP20CheetahVisualization
_ZN25MiniCheetahHardwareBridge10runUdpRecvEv
_ZN25MiniCheetahHardwareBridge10runUdpSendEv
_ZN25MiniCheetahHardwareBridge11runWatchDogEv
_ZN25MiniCheetahHardwareBridge12runTpCamRecvEv
_ZN25MiniCheetahHardwareBridge14runHighCmdRecvEv
_ZN25MiniCheetahHardwareBridge15runAppStateSendEv
_ZN25MiniCheetahHardwareBridge16runHighStateSendEv
_ZN25MiniCheetahHardwareBridge3runEv
_ZN25MiniCheetahHardwareBridgeC1EP15RobotControllerb
_ZN25MiniCheetahHardwareBridgeC2EP15RobotControllerb
_ZN25MiniCheetahHardwareBridgeC4EP15RobotControllerb
_ZN25MiniCheetahHardwareBridgeD1Ev
_ZN25MiniCheetahHardwareBridgeD2Ev
_ZN25MiniCheetahHardwareBridgeD4Ev
_ZN26cheetah_visualization_lcmt11getTypeNameEv
_ZN26cheetah_visualization_lcmt12_computeHashEPK15___lcm_hash_ptr
_ZN26cheetah_visualization_lcmt13_decodeNoHashEPKvii
_ZN26cheetah_visualization_lcmt6decodeEPKvii
_ZN26cheetah_visualization_lcmt7getHashEv
_ZNK26cheetah_visualization_lcmt13_encodeNoHashEPvii
_ZNK26cheetah_visualization_lcmt14getEncodedSizeEv
_ZNK26cheetah_visualization_lcmt21_getEncodedSizeNoHashEv
_ZNK26cheetah_visualization_lcmt6encodeEPvii
_ZTI22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeE
_ZTS22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeE
_ZTV22PeriodicMemberFunctionI25MiniCheetahHardwareBridgeE
_mainCheetahVisualization
buildCheetah3<float>
buildMiniCheetah<double>
buildMiniCheetah<float>
cheetah
cheetahMainVisualization
cheetah_visualization_lcmt
cheetah_visualization_lcmt.hpp
setMaxTorqueCheetah3
~MiniCheetahHardwareBridge
```

mbleviathan, 2022-08-22:
Seconded.

现在不是因为一个简单的文书错误而“丢脸”的时候。 MIT 许可证很容易遵循！

我们的目的不是让您难堪，我们的目的是更正一个小错字并让您的代码符合许可证。

martinbogo, 2022-08-22:
Seconded.

现在不是因为一个简单的文书错误而“丢脸”的时候。 MIT 许可证很容易遵循！

我们的目的不是让您难堪，我们的目的是更正一个小错字并让您的代码符合许可证。

MAVProxyUser, 2022-09-10:
If also you feel like sharing the source for sportMode, that would be glorious.

MAVProxyUser, 2022-09-23:
Legged_sport is clearly referencing the MIT Cheetah HardwareBridge

https://mit-biomimetics.github.io/d9/da0/_hardware_bridge_8cpp_source.html
https://mit-biomimetics.github.io/d5/d90/_control_parameters_8cpp.html
https://github.com/mit-biomimetics/Cheetah-Software/blob/master/common/src/Controllers/FootSwingTrajectory.cpp

Will you share so that it can be modified please?