unitreerobotics/unitree_legged_sdk #57: Go1-- no action

wongearth, 2022-09-06:
I built the unitree_legged_sdk ver 3.8.0 successfully... trying to run provided examples  
sudo ./example_walk. 
The programs seems to be running but the robot does not response at all. No errors reported.. nothing!!  I ping the robot both at 192.168.123.161  and 192.168.123.10 and the connections are ok. What do I miss? Is there button at the joystick to set the mode that accept command from pc? Please help. 

My system info:
ubuntu 18.04
unitree_legged_sdk 3.8.0

SimonGodon, 2022-11-11:
I had the same issue, try with a lower version of the unitree_legged_sdk. I tried with v3.4.2 and it worked.

JonasFovea, 2022-11-28:
[This forked version](https://github.com/JonasFovea/unitree_legged_sdk/tree/fix) is working for me.

MAVProxyUser, 2022-12-18:
Stop by and see us on Slack, we will try to help you. 
https://join.slack.com/t/robotdogs/shared_invite/zt-1fvixx89u-7T79~VxmDYdFSIoTnSagFQ