unitreerobotics/unitree_legged_sdk #58: Go1 motor angle feedback inaccurate

echoGee, 2022-09-19:
Is the angle feedback returned by the motors in this message 
https://github.com/unitreerobotics/unitree_ros_to_real/blob/master/unitree_legged_msgs/msg/MotorState.msg based on an absolute encoder or relative encoder ? 
If its relative, what is q=0 initialized as ? 
The angle reported seem to be different after every boot.

JonasFovea, 2022-12-06:
I can only give you a guess, but I think it's relative encoding, which is set to zero at the startup of the robot.

We once angled a hind leg slightly while starting the robot, which resulted in the leg being completely off and therefore the robot being unbalanced and so on...

xyq1223, 2023-03-13:
Hello,our robot dog uses a single-turn absolute encoder

JonasFovea, 2023-03-27:
@xyq1223 

But that doesn't correspond to the behaviour of the robot.
If it was absolute, the behaviour shouldn't be determined by the motor positions at startup...