unitreerobotics/unitree_legged_sdk #60: python examples for A1?

pab47, 2022-09-30:
Are there python examples for A1 (e.g., v 3.3.1) ? It would be nice to use A1 via python code. 

I know that there is python examples for Go1 in v 3.8.0

JonasFovea, 2022-12-06:
My guess is, that they're not going to implement more features for the A1 since the Go1 is its predecessor...

But you could write your own cpython wrapper for the SDK, since the header files are known and you have the static library...

If that is too much work for you, I would suggest to use the unitree_ros_to_real package to interface the robot via ROS.
This way, you can send ROS messages from Python, which then get translated to the otherwise used UDP messages.

xyq1223, 2023-03-13:
Sorry we don't provide python examples for A1, you can refer to Go1 pybind11
https://github.com/unitreerobotics/unitree_legged_sdk/tree/go1/python_wrapper

Maddy1206, 2023-05-31:
@pab47 
I wrote a python wrapper similar to the one provided by the Go1. 
Unfortunately, due to firmware issues I was not able to use the v3.3.1 on my A1, therefore I stuck to the v3.2. 
You can find it on my profile.