unitreerobotics/unitree_legged_sdk #61: 3.8.0 Python examples not reading UDP state information

MulkerinG, 2022-10-04:
When running the default python examples, the information from the received state is being read as all zeros. Originally I just uncommented the given print functions in the code and all motionstate variables were zero, to make sure this wasn't simply a motionstate object problem I also added the line for the imu temperature.  The following image is the results. 

[image]

The Robot is running the commands from the script fine but I would like to use some of this information for logging purposes. Is there a setting I am missing or is this not currently possible with the python examples?

JonasFovea, 2022-11-28:
How are you connected (Wi-fi/Ethernet) and which IP are you using to address the robot?

kg0221, 2023-07-12:
Has this been resolved? I'm seeing the same thing using a wired connection.