unitreerobotics/unitree_legged_sdk #62: Have any version support B1.

Cmccm123, 2022-10-17:
We allready buy a B1 and want to use sdk to control B1. However, we can't find any version of 
unitree_legged_sdk which support B1. We find a unitree_legged_sdk on xavier, but the version is 3.8. We don't know 3.8 can support B1(refer to README.md)

Have any version support B1?

JonasFovea, 2022-12-06:
Hi!

In case you did not check the repo frequently, there's now a version for B1:
https://github.com/unitreerobotics/unitree_legged_sdk/tree/v3.8.3

JonasFovea, 2022-12-06:
I'd love to hear, if the version works, since the version naming and the support for different Robots seems to be out of control since v3.8.1 ...
They should have made branches for each robot individually