unitreerobotics/unitree_legged_sdk #63: [Urgent] IP & Port Error when running example files in A1

tlangdun, 2022-10-26:
Robot: A1
Nvidia board IP: 192.168.123.12
Pi Ip (rear end of the robot): 192.168.123.161

Steps to recreate:
1. Connected to Nvidia board
2. Run sudo ./example_walk
3. Error pops up: Error: Bind client ip&port failed.

Do I have to configure my ip and port to solve this issue? How should I proceed?

JonasFovea, 2022-12-06:
Are the ports already in use?
Is the IP configured correctly?

Which version of the SDK are you using?

JonasFovea, 2022-12-07:
I have heard, that some versions may not work due to outdated firmware.
Since the ports you described have changed, I would guess, that your current firmware doesn't support the SDK version.

@TrivasZhang might be able to confirm that

You'd probably need to specify the firmware version.

For now I would suggest to stick with the older SDK version, if that works for you.

tlangdun, 2022-12-07:
Thank you for the information!

Maddy1206, 2023-05-31:
@tlangdun 
Maybe you can run the v3.3.1 on the Pi. Worked for me, but on the NX it did not work, too.

guyo-shifters, 2024-02-07:
this helps a lot :)
since then did you manage to run it with anything newer then 3.2? I am having the same problem and would like to use 3.3 high level functions