unitreerobotics/unitree_legged_sdk #64: HighState isn't sent in every network configuration

JonasFovea, 2022-10-28:
I have spotted a problem with the return of the HighState, when connected via WiFi.

# Current behavior

Currently a difference is made if you are connected via LAN or WiFi.
If you connect via LAN, you direct the UDP messages to `192.168.123.161` and if you are connected via WiFi, you direct the UDP messages to `192.168.12.1`.

Therefore switching between LAN and WiFi forces you to change the IP in your code and recompile it (which is obviously not what you want, if you switch it regularly).

It is further possible to [set the Raspberry Pi into network pass through mode](https://youtu.be/tTCbdul7xsc?t=454) to send IP messages from the `192.168.12.0/24` network to the `192.168.123.0/24` network. For this, a route to the internal network of the robot has to be configured, in your connecting PC,  via the Raspberry Pi (either as default gateway or static route).

If this is set up, you are able to connect to all devices in the `.123.0` network, which also includes the Pi's interface with address `192.168.123.161`.
You are now able to control the Go1 robot via WiFi by directing the UDP messages to `192.168.123.161`, which is great.

## The Problem
When doing so, you don't get HighState messages via UDP.

# Expected behavior
Since the network connection works in both ways, the HighState should be returned via UDP.

HBSIM0, 2022-11-16:
Try 192.168.12.1

JonasFovea, 2022-11-16:
@HBSIM0 that's what I currently do.
But having to change it manually every time you switch between wired and wireless is quite error-prone.

I would rather always set the IP to the 161 address and configure the forwarding once and for all.

xyq1223, 2023-03-13:
Sorry, our SDK does not support cross-network segment communication