unitreerobotics/unitree_legged_sdk #66: Unable to send commands from external PC [Go1 Edu]

SimonGodon, 2022-11-14:
Hello, 

I have a problem with unitree_legged_sdk versions 3.5.1 and 3.8.0. I try to run the 4 examples (walk, position, velocity, torque) but nothing happens, and the terminal prints only lines of zeroes. 
When I do exactly the same thing using the unitree_legged_sdk v3.4.2, it works very well!
But I have to use a later version to use the unitree_ros_to_real which requires 3.5.1, so I can't keep using 3.4.2.
 
I am connected with ethernet, and the IP address is correct, connection works. 

Here is what I found: 
By replacing files from the newer version (3.5.1) with files from the old one (3.4.2), I noticed that nothing changed, until I replaced libunitree_legged_sdk_amd64.so. At this point, finally the robot moved and the values in the terminal were not zeroes anymore. The problem is that the leg of the robot went abruptly to the max range and stayed there without moving. 

So in short: what in the newer versions make it impossible for the robot to move and receive commands correctly? Seems like it has something to do with the '.so' file but I cannot investigate further.

Thank you!

SimonGodon, 2022-11-15:
Should I update the firmware maybe? That's the only solution I can think of, but it's not recommended according to Unitree (https://m.unitree.com/go1_update/)

MAVProxyUser, 2022-12-18:
SDK versions are very specific for Go1. 
https://github.com/MAVProxyUser/YushuTechUnitreeGo1/blob/main/SDK.md

anahrendra, 2023-03-23:
Hi! I am experiencing a slightly similar problem.

I can send commands to the robot, but somehow the ethernet connection is unstable. It can be turned off suddenly. I made sure that the LAN cable is okay. Is this due to IP problem?

I made a static connection to 192.168.123.115 for connecting from external PC to the robot's ethernet port.
And in the SDK, I set `udp(LOWLEVEL, 8090, "192.168.123.10", 8007)`

Thanks in advance for your help!

anahrendra, 2023-06-06:
> Hi @anahrendra , we seem to have the same problem. The robot makes a clicking noise and shuts off sometimes as soon as we pass a non-zero command via UDP in low-level. It only happens when using the ethernet cable. Did you find a way to fix it?

Hi,

in our case, we solved it by making sure the robot enters low-level mode by pressing `L1+A, L1+A, L2+B, L1+L2+Start` from the controller