unitreerobotics/unitree_legged_sdk #67: Compatibility with non-edu version

matteoerrera, 2022-11-22:
Hi, I am planning to purchase a Unitree Go 1 Air, but from what I understand this SDK is not compatible with non-edu products.
Are there any alternatives to be able to communicate with the robot? Are there plans to extend compatibility to non-edu models as well?

Thank you

JonasFovea, 2022-11-28:
That's their business model ;)
You pay extra for the software interface...

MAVProxyUser, 2022-12-18:
This is what you need here: 
https://github.com/MAVProxyUser/YushuTechUnitreeGo1/blob/main/SDK.md

Feel free to stop by and see us here for help using the SDK with non EDU models. 
https://join.slack.com/t/robotdogs/shared_invite/zt-1fvixx89u-7T79~VxmDYdFSIoTnSagFQ