unitreerobotics/unitree_legged_sdk #69: !! Version confusion !!

JonasFovea, 2022-11-28:
Hey all together!

I am really surprised about the recent wave of commits by @Dyingisasnaturalasliving, which changed versions back to Aliengo support.

Make sure to select your version carefully.
Go1 users should at least fall back to the [last commit on version 3.8.0](https://github.com/unitreerobotics/unitree_legged_sdk/tree/7a09517e6398b36501740d6ca0cc5dc052890666), if not even further.

MAVProxyUser, 2022-12-18:
As I have documented here: https://github.com/MAVProxyUser/YushuTechUnitreeGo1/blob/main/SDK.md

Do not use one of these versions of the SDK, because they do NOT support go1: https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/v3.2 https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/v3.3 https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/v3.3.1 https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/v3.3.2 https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/3.3.3 https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/3.3.4 https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/v3.8.1 https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/v3.8.2 https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/v3.8.3 https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/v3.8.4

realquantumcookie, 2023-05-15:
Isn't the Go1 branch newer than 3.8.0?

JonasFovea, 2023-05-15:
At the time it wasn't
I haven't checked again, if something changed

xyq1223, 2023-05-16:
Hello, the latest version of our Go1 SDK is 3.8.0

JonasFovea, 2023-05-16:
But what about 3.8.6?
It states Go1 support
@xyq1223

realquantumcookie, 2023-05-16:
> But what about 3.8.6? It states Go1 support @xyq1223

+1 for this

xyq1223, 2023-05-16:
Sorry for the mistake in the reply just now, the latest version is 3.8.6