unitreerobotics/unitree_legged_sdk #74: Lib for python 3.10 (Go1 EDU)

Kulczar, 2022-12-27:
Hello guys! 

the lib works very well on python 3.8, i want to know if able to upgrade the lib to works on python 3.10 !?

Thx! 

https://github.com/unitreerobotics/unitree_legged_sdk/tree/v3.8.0

JonasFovea, 2022-12-30:
Doesn't it work with 3.10?

Kulczar, 2023-01-04:
No :( just with python 3.8!
[image]

MiSTerFPGA, 2023-03-10:
@Kulczar Consider using Free-Dog-SDK instead? https://github.com/Bin4ry/free-dog-sdk since Unitree support is so awful, many of us are banging together on Slack and learning to get around their limitations. 
Stop on by! https://join.slack.com/t/robotdogs/shared_invite/zt-1fvixx89u-7T79~VxmDYdFSIoTnSagFQ

xyq1223, 2023-03-13:
Hello,we have provided the python_wrapper,please check and any version of Python （>3.0） can be used to control the robot dog
(https://github.com/unitreerobotics/unitree_legged_sdk/tree/go1/python_wrapper)

JonasFovea, 2023-03-27:
@Kulczar have you tried the new pybind feature?

realquantumcookie, 2023-04-02:
> Hello,we have provided the python_wrapper,please check and any version of Python （>3.0） can be used to control the robot dog (https://github.com/unitreerobotics/unitree_legged_sdk/tree/go1/python_wrapper)

Hi there, just follow up on this, I tried to compile for python 3.11 but it failed. Looks like the newest supported version of python for the pybind feature you guys provided would be 3.10. And also inside [python_wrapper/python_interface.cpp](https://github.com/unitreerobotics/unitree_legged_sdk/blob/go1/python_wrapper/python_interface.cpp) it looks like the line

```
#include <msgpack.h>
```

is actually not needed.

closed 2023-05-04 by Kulczar