unitreerobotics/unitree_legged_sdk #77: Build python wrapper for other python versions

IrisLi17, 2023-01-08:
Hi, 
The python wrapper in v3.8.0 works perfectly with python 3.8 on Go1, but I am wondering if it is possible to open-source the wrapping and building code so that one can build python wrappers for other python versions (e.g., python 3.6). Thanks!

Wsser, 2023-03-07:
We will open source this wrapper method soon in the next few days.

MiSTerFPGA, 2023-03-10:
@RbShine 
Unitree support is awful. Stop by and see us on slack instead, where the community struggles together, and learns from each other. 
https://join.slack.com/t/robotdogs/shared_invite/zt-1fvixx89u-7T79~VxmDYdFSIoTnSagFQ

@IrisLi17 try the "Free-dog-SDK" instead. You don't need their bloated software bundle in most cases. 
https://github.com/Bin4ry/free-dog-sdk

IrisLi17, 2023-03-10:
@Wsser  @MiSTerFPGA Thanks but I have already implemented a wrapper with pybind11, much similar to the one in motion_imitation (https://github.com/erwincoumans/motion_imitation/blob/master/third_party/unitree_legged_sdk/python_interface.cpp)

xyq1223, 2023-03-13:
Hello,we have provided this python_wrapper,please check
(https://github.com/unitreerobotics/unitree_legged_sdk/tree/go1/python_wrapper)

closed 2023-03-13 by IrisLi17