unitreerobotics/unitree_legged_sdk #84: catkin_make bug

RbShine, 2023-03-08:


RbShine, 2023-03-08:
When i put it in catkin_ws/src and catkin_make, it returns this:
`/home/liu/catkin_ws/src/unitree_ros_to_real-master/unitree_legged_real/src/exe/twist_sub.cpp:6:10: fatal error: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory
    6 | #include "unitree_legged_sdk/unitree_legged_sdk.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
unitree_ros_to_real-master/unitree_legged_real/CMakeFiles/twist_sub.dir/build.make:62: recipe for target 'unitree_ros_to_real-master/unitree_legged_real/CMakeFiles/twist_sub.dir/src/exe/twist_sub.cpp.o' failed
make[2]: *** [unitree_ros_to_real-master/unitree_legged_real/CMakeFiles/twist_sub.dir/src/exe/twist_sub.cpp.o] Error 1
CMakeFiles/Makefile2:3260: recipe for target 'unitree_ros_to_real-master/unitree_legged_real/CMakeFiles/twist_sub.dir/all' failed
make[1]: *** [unitree_ros_to_real-master/unitree_legged_real/CMakeFiles/twist_sub.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`
i checked the REAADME of ros_to_real package, it says that
`## Dependencies
* [unitree_legged_sdk](https://github.com/unitreerobotics/unitree_legged_sdk/releases)
### Notice
The newest release [v3.8.0](https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/3.8.0) only supports for robot: Go1.

Check release [v3.3.4](https://github.com/unitreerobotics/unitree_legged_sdk/releases/tag/3.3.4) for A1 support.`

so the version of two package is supposed to be right.i don't know why it returns a bug.
Please help me.

MiSTerFPGA, 2023-03-10:
@RbShine 
Unitree support is awful. Stop by and see us on slack instead, where the community struggles together, and learns from each other. 
https://join.slack.com/t/robotdogs/shared_invite/zt-1fvixx89u-7T79~VxmDYdFSIoTnSagFQ

Try following these instructions here instead. 
https://github.com/MAVProxyUser/unitree_ros2_ws

Even though you don't pay our slack group ungodly amounts of money for support, we will do our best, and at times better than you'll ever get with Unitree support team.

xyq1223, 2023-03-13:
1.Please pay attention to whether the SDK path is under the src path
2.Also, please note that the SDK name and path in ros_to_real are correct
https://github.com/unitreerobotics/unitree_ros_to_real/blob/c54489e721ee88d3b6851a44fc79ae75fc0583bb/unitree_legged_real/CMakeLists.txt#L30