unitreerobotics/unitree_legged_sdk #90: Build Error: Go1

rafiqrana, 2023-04-22:
I am getting the following error during the build process:

Build Error:
```/usr/bin/ld: /home/<username>/git/clone/unitree_legged_sdk/lib/cpp/amd64/libunitree_legged_sdk.a(loop.cpp.o): relocation R_X86_64_PC32 against undefined hidden symbol `_ZTCN5boost10wrapexceptINS_17bad_function_callEEE0_NS_16exception_detail10clone_implINS3_19error_info_injectorIS1_EEEE' can not be used when making a PIE object
/usr/bin/ld: final link failed: bad value```

System info:
```
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy
```

xyq1223, 2023-04-28:
Hello,we have fed back this problem to our R&D colleagues, and they will reproduce the problem, please look forward to it

tfwittwer, 2023-05-17:
Same problem here, also on Ubuntu 22.04. Boost version is 1.81.0 compiled from source. I think the cause is that libunitree_legged_sdk.a was not compiled with the -fPIC option.

xyq1223, 2023-05-17:
Hello, this is a problem with the compilation of Ubuntu22.04

NOTMOVETON, 2023-06-07:
Hello, have this problem been solved? if so, how to remove this error

xyq1223, 2023-06-07:
Hello, we do not support the solution of the version of Ubuntu itself for the time being. It is recommended that you can choose to use Ubuntu18.04.

sharathmatada, 2023-08-16:
Hi, 
Is this issue resolved for Ubuntu 22.04?

ahanjaya, 2024-06-03:
Hi, here is my way to compile the sdk with ubuntu 22:
1. Download the older [libboost-1_66 ](https://nchc.dl.sourceforge.net/project/boost/boost/1.66.0/boost_1_66_0.7z?viasf=1)
2. unzip the downloaded libboost
3. ./bootstrap.sh
4. ./b2

then edit the CMakeLists.txt like following:
```
cmake_minimum_required(VERSION 2.8.3)
project(unitree_legged_sdk)

# Set Boost directory
set(BOOST_ROOT "/directory_to_the_compiled_boost_1_66_0")

# Add Boost include directory to compiler include paths
include_directories(${BOOST_ROOT})

# Add Boost library directory to linker library paths
link_directories(${BOOST_ROOT}/stage/lib)
```

 redo the `cmake ..` and `make` the sdk folder.