unitreerobotics/unitree_legged_sdk #91: Go1: Demos in example_py folder do not work

rafiqrana, 2023-04-22:
Hi, I am getting the following error with the go1 branch:

```
$ python3 example_py/example_walk.py 
Traceback (most recent call last):
File "/home/<username>/git/clone/unitree_legged_sdk/example_py/example_walk.py", line 8, in <module>
    import robot_interface as sdk
ModuleNotFoundError: No module named 'robot_interface'
```

System info:
```
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy
```

xyq1223, 2023-04-28:
Hello,We recommend that you download the latest SDK and recompile and run

sxyxs, 2023-11-03:
Hi I download the latest SDK (3.8.6 for go1),  compile and run (ubuntu20.04) with ```cmake -DPYTHON_BUILD=TRUE .. ```

I still got the   issue ```python3 example_py/example_walk.py 
Traceback (most recent call last):
  File "example_py/example_walk.py", line 8, in <module>
    import robot_interface as sdk
ModuleNotFoundError: No module named 'robot_interface'```

heartInsert, 2024-07-05:
Do you solve this  problem ?