unitreerobotics/unitree_legged_sdk #94: Go1 Minimum Command Update Frequency

dasnyder5, 2023-05-10:
When running the Go1 with lidar for obstacle detection and localization, we observe the following behavior when modifying example_walk.py (HighLevel control) file: (Note: frequencies here refer to the inverse of the sleep interval dt)

For frequencies around 10-20Hz, there is essentially no motion
For frequencies around 50Hz, there is sporadic motion
For frequencies around 100Hz, there is pretty good (mostly smooth) motion
For frequencies > 200Hz, there is smooth motion. 

When using the Lidar for obstacles only (no localization), the interference issue doesn't appear (i.e. commands sent at 10Hz work fine). In looking through the source code, there is a comment on the sleep command in the example_walk.cpp file noting that the sleep dt should be in 0.001-0.01 (i.e., operate at 100-1000Hz). In the hopes that this means the developers also observed this phenomenon, I wanted to ask:

Can the reasons for this observed interference (and the 100-1000Hz comment) be clarified? Is there an easy way to hack together a solution so that commands issued at, say, 10 Hz will also work when localization is running?

xyq1223, 2023-05-18:
Hello, the scene description of the problem you raised is quite special, and the details are not enough, please provide further relevant detailed description and screenshots, so that we can make further judgments, thank you

dasnyder5, 2023-05-18:
Hi, 

We think we found the issue. It appears that the updateCommand() within ros_control was updating at a rate of AX_CONTROL_PERIOD=0.005; changing this variable appears to have fixed the problem. Thank you for your response!

closed 2023-05-18 by dasnyder5