unitreerobotics/unitree_legged_sdk #95: compilation error with libboost >= 1.74.0 and GO1 libunitree_legged_sdk.a

VModugno, 2023-05-21:
Hi all,

I'm writing you because I have found that unitree_legged_sdk does not link appropriately if the libboost installed on the machine is equal to or higher than 1.74.0. 
the error that I get when I compile with libboost 1.74.0 and 1.78.0 is: 
`libunitree_legged_sdk.a(loop.cpp.o): relocation R_X86_64_PC32 against undefined hidden symbol `_ZTCN5boost10wrapexceptINS_17bad_function_callEEE0_NS_16exception_detail10clone_implINS3_19error_info_injectorIS1_EEEE' can not be used when making a shared object
  /home/robohikeuser/mambaforge/envs/hori/compiler_compat/ld: final link failed: bad value`
  as soon I force during the compilation of the examples with the cmakelist.txt with an older version of libboost (1.71.0) everything compiles perfectly and works on the robot.
  Therefore it would be great if you could provide a version of GO1 libunitree_legged_sdk.a that has been compiled with a more recent version of libboost (like 1.78.0). This issue is preventing me to use your library on more recent Ubuntu versions or with the robostack project under conda/mamba.
  
thank you very much for your help

xyq1223, 2023-05-22:
Hello, about the problem you mentioned, can you please align the version of libboost with our side

VModugno, 2023-05-22:
dear @xyq1223 

what do you mean by that? Of course if I rollback the libboost version it works but It would be great if you can ship the library for more recent versions of libboost
thank you for your help

VModugno, 2023-05-22:
thank you for your reply and thank you for your support

sharathmatada, 2023-08-16:
Hi,
Were you able to build the package in Ubuntu 22.04 with older versions of libboost?

VModugno, 2023-08-31:
I'm currently using your sdk with pybind because i made my own version of a python wrapper, as far as i know i have been able to compile the pybind on my machine (which has ubuntu 22) as long as i was using a version of boost compatible with your sdk. 
I do perform the compilation under a conda environment with the right libboost version. I can do some checks if needed

agoeckner, 2024-03-26:
Hello, is there any update on this issue?

ahanjaya, 2024-06-03:
Hi, here is my way to compile the sdk with ubuntu 22:
1. Download the older [libboost-1_66 ](https://nchc.dl.sourceforge.net/project/boost/boost/1.66.0/boost_1_66_0.7z?viasf=1)
2. unzip the downloaded libboost
3. ./bootstrap.sh
4. ./b2
5. sudo ./b2 install

 redo the `cmake ..` and `make` the sdk folder.