unitreerobotics/unitree_legged_sdk #96: Friction compensation in the low level controller

Sergim96, 2023-05-22:
Hi,

We are experiencing quite a lot of friction in the actuation of the motors on B1, specially on the knee one. Does the low level controller have some kind of friction compensation implemented? If so, how can it be enabled?

This is a key component when performing torque control in any robotic system, however, the lack of documentation makes it difficult to know if this feature exists or not.

xyq1223, 2023-05-24:
Hello, there is no friction compensation, it is recommended that you can adjust it slowly. In addition, the mass of the imu is very small and has no effect on the dynamic model

zixuan417, 2023-10-22:
Hi, we also encountered a similar problem. Could you kindly tell me how you fixed this?