unitreerobotics/unitree_legged_sdk #103: Protection features in safety.h

anahrendra, 2023-06-05:
Hi, I have some concern about the safety features in the SDK.

In `safety.h`, there is a `PowerProtect` function that limits the power in the given low-level command.

What is the default value of this function? If I don't explicitly implement this function in my code, does the SDK set the `PowerProtect` function to a certain level?

Thanks for your help!

xyq1223, 2023-06-07:
Hello, first of all, this function must pass parameters. And we do not support providing default protection values in code (only hardware will provide default protection)

anahrendra, 2023-06-07:
Hi, thanks for your reply.

That means, if I pass the parameters using this `PowerProtect` function in my code, (e.g., with parameter=8), then it will overwrite the hardware's default protection using my given parameter. Am I right?

Thanks!

xyq1223, 2023-06-09:
Hello,in the example you mentioned, for example, the parameter is set to 8 in the code, which means that you assign a value to the protection on the software. But the protection on the hardware I mentioned earlier refers to the default factory value when the electronic components are produced.

anahrendra, 2023-06-23:
> Hello,in the example you mentioned, for example, the parameter is set to 8 in the code, which means that you assign a value to the protection on the software. But the protection on the hardware I mentioned earlier refers to the default factory value when the electronic components are produced.

Hi again. I keep experiencing issues with the Low Level control. Even when I disable or increase the level to 9, the robot still sometimes get its legged disabled. Is there anything like torque protection mode in the SDK? It seems that it mostly happens when the motor cannot track the given position target for a long time.

Thanks!

xyq1223, 2023-08-03:
Hello, if the phenomenon of disabled occurs frequently, it is recommended not to use the protection function. You can write a software limit yourself. Once the limit you set is exceeded, send zero torque zero kp kd to the motor to protect the motor