unitreerobotics/unitree_legged_sdk #104: How to get Low-Level Control Working

andrewipstirling, 2023-06-08:
Hi,

Very confused on the documentation provided for low-level control. High-Level control examples work but no low-level ones do. Any help would be very appreciated. Have already tried:
- Setting levelFlag = 'LOWLEVEL'
- Setting cmd.head[0] and [1] to 0xfe and 0xef respectively
- changing port numbers

We are on legged_sport version 1.38. We are in low-level programming mode (L2 + A, L2 + A, L2 +B, L1 + L2 + START).

Thanks

xyq1223, 2023-06-13:
Hello, we presume that your robot dog is not switched to normal mode. If the dog makes a beeping sound when running, try switching the mode again

yiyuc, 2023-07-26:
You can ssh into the raspberry PI board and then kill the legged_sport process. Then the robot is 100% in low-level mode

xyq1223, 2023-08-04:
Hi, yes, to make sure the robot dog is in low-level mode, ssh into the raspberry pi board, then kill the legged_sport process.

andrewipstirling, 2023-08-04:
Killing the legged_sport_process doesn't fix it. I get data back but cannot send any commands to dog.

xyq1223, 2023-08-07:
Hello, is your robot dog a Go1 edu (only edu models are suitable for this method)

Yaru-Gu, 2024-02-08:
Hello @xyq1223, will (L2 + A, L2 + A, L2 +B, L1 + L2 + START) put the robot Go1 EDU into the normal mode?