unitreerobotics/unitree_legged_sdk #106: example_joystick with python

Qianzhong-Chen, 2023-06-30:
Hi, I found that example_joystick only has cpp version but no python program.

I am using python to write control code and want to use the RC at the same time.  However, when using python to collect data from RC, the data in state.wirelessRemote doesn't make sense to me. Then I found that in example_joystick.cpp, it uses memcpy, which is a feature in cpp. 

Could you please develop a python version example_joystick program? That would be super helpful to me, thanks!

xyq1223, 2023-08-03:
Hello, we have fed back this requirement to our R&D colleagues, thank you for your support and understanding！

Morozovga691, 2023-09-08:
Could you tell me please, what example_joystick program should do? When I try to start the file, i see the next problem: " ./example_joystick: error while loading shared libraries: libunitree_legged_sdk.so: cannot open shared object file: No such file or directory "