unitreerobotics/unitree_legged_sdk #110: Follow mode setting

JL-Brain, 2023-08-08:
Hello, I'm trying to control go1 robot with ROS.

I read the provided manual's explanation, but I still have a question.
Is there any way to start and stop human following mode without mobile APP?
I want to know python or c++ code to control it

Thanks

JL-Brain, 2023-08-10:
Can I control robot dog's posture when robot dos is follow mode?

xyq1223, 2023-08-14:
Hello,first of all, the following mode of the robot dog must be started and stopped through the mobile phone APP, and secondly, the command of the Label tracking mode is transmitted from the APP to the robot dog through mqtt.

JL-Brain, 2023-08-18:
How does the app receive the MQTT signal? 
What is the IP address and port number?

JL-Brain, 2023-08-18:
Thanks,
Can you provide
"/home/pi/Unitree/autostart/utrack/catkin_utrack/src/utrack/a2_ukd/src/uwb_t
riple_udp_node.cpp"

this file to me?

xyq1223, 2023-09-27:
Hello, sorry we do not currently provide uwb files