unitreerobotics/unitree_legged_sdk #111: Position control without output torque command

dhruvkm2402, 2023-08-17:
Hello,
I was changing parameters of the example_py position control script for testing. What I noticed is that for the hip joint, without inputting desired output torque it does not achieve the required angle. In the script of sine wave movement, for hip joint there is a desired output torque of -0.65 but 0 for others. Is it not possible to do a pure position control without estimating the desired output torque?

Charlescai123, 2023-11-17:
Yes, facing the same issue and wish to know how to resolve it

Charlescai123, 2023-11-25:
You need to set proper Kp and Kd inside the commands for each leg. The reason why it cannot achieve the desired position is probably because Kp/Kd are set pretty small. To make it work, identify suitable values for the parameters