unitreerobotics/unitree_legged_sdk #113: How to use access HighState while doing low level control?

sharathmatada, 2023-08-29:
Hi,
I'm trying to do low-level control but would like to use the in-built state estimator and would like to access the variables in HighState to perform low level control. Any leads on this will be helpful

heartInsert, 2024-07-04:
@xyq1223    mate  ,  do  you  know  how to get /gazebo/model_states    and  /aliengo_gazebo/joint_states    in  a  real  robot ?

alvarobelmontebaeza, 2024-11-20:
Hi! Is there a way to get the estimated state while in low level mode? Or access the state estimator separately and run it ourselves while in low state mode? Why can't the state estimator keep working while in lowlevel control mode?

heartInsert, 2024-11-21:
> Hi! Is there a way to get the estimated state while in low level mode? Or access the state estimator separately and run it ourselves while in low state mode? Why can't the state estimator keep working while in lowlevel control mode?

what is   estimated state ?

alvarobelmontebaeza, 2024-11-21:
> > Hi! Is there a way to get the estimated state while in low level mode? Or access the state estimator separately and run it ourselves while in low state mode? Why can't the state estimator keep working while in lowlevel control mode?
> 
> what is estimated state ?

When running the SDK in HighLevel mode, one can access to an estimate of the robot body position and velocity, which I assume is inferred by a built-in state estimator that leverages the onboard sensors of the robot as well as maybe leg kinematics. The problem is that when switching to LowLevel mode, you  cannot access that information anymore, which doesn't make sense to me because state estimation should not be coupled to the joint commands outputed by the built-in controller in highlevel mode.

heartInsert, 2024-11-21:
@alvarobelmontebaeza  Oh , do you  mean  you  want  to  get  robot's   body  position and velocity in  low level ?   It's  definitely  cound ,  because  I  used  to   access  these  information  by  Ros   ,   and  I  run  my  RL  model  in  real  robot   in  low_level  control   successfully.

alvarobelmontebaeza, 2024-11-21:
@heartInsert Are you sure? I mean, the reason for this topic is issue is that the estimated state is not accessible in LowLevel mode. When you set the robot into lowlevel mode, it stops updating the values of the highLevel state message. 

If it is working for you, could you share some code?

heartInsert, 2024-11-21:
@alvarobelmontebaeza   I  don't  know  what  messages you  want  ,  I  remember  that  I   can  get   robot's  12  joint's  position and  acceleration  and  orientation ,  all  of  these  information  is  updating  all the time ,  do you want these ?   I  didn't  run  my  robot  in  High-level , so  I   don't  know the   information  in  High-level.

alvarobelmontebaeza, 2024-11-21:
That information is joint-level information and IMU sensor raw info, which is quite noisy and you cannot really rely on that. If you run in HighLevel mode, you get also an estimate of global position and velocity of the body, which is what I'm interested in. Maybe @xyq1223 who replied earlier in this thread has more insight

mcamurri, 2025-05-29:
I second that request. Support from local unitree distributors told me it is not possible though. At least not on Aliengo.

dipampatel18, 2025-09-18:
@alvarobelmontebaeza Were you able to find any solution? I too need to access the body velocity from the HighState struct. But all I receive is zeros.