unitreerobotics/unitree_legged_sdk #118: About libunitree_legged_sdk.so

muye1202, 2023-10-03:
Hi!

May I ask if it is possible to make public the dynamic version of libunitree_legged_sdk? I'm trying to compile a cpp script that's using unitree_legged_sdk library, and has encountered problem when using the static version of libunitree_legged_sdk.It's giving me undefined symbol error and I'm using CMake as compilation tools.

My work is NOT for any commercial use, I'm currently doing research in a robotics lab at Northwestern University in the U.S.

thanks!

Liu-Xiaoyan97, 2023-11-08:
Have you solved your problem? I ran into a similar problem. I renamed robot_interface* in lib/python/arm64 in the sdk to robot_interface.so and placed it in the same directory as the script. It did import, but I encountered ImportError: undefined symbol:PyInstanceMethod_Type. How did you solve it? thank you

giuseppecialdella, 2023-11-08:
@Liu-Xiaoyan97 Not yet, I've tried what you are saying but it gives me another error: **ImportError: libunitree_legged_sdk.so: cannot open shared object file: No such file or directory**