unitreerobotics/unitree_legged_sdk #124: SDK

Mystic4l, 2023-10-25:


Mystic4l, 2023-10-25:
Hello,
I have downloaded the sdk v3.8.0 for the go1.But when I failed to make.
[image]
Does anyone know how to handle the problem? Thank you in advance!

Jialiangfan1997, 2024-01-30:
我遇到了一样的问题，请问你是如何解决的，谢谢～

Jialiangfan1997, 2024-01-30:
HI, I met the same problem. Could you please tell me hold to solve it ?  Thanks!

Jialiangfan1997, 2024-01-30:
> Hello, I have downloaded the sdk v3.8.0 for the go1.But I failed to make. ![image](https://private-user-images.githubusercontent.com/127591572/277971204-6c1171de-9911-4268-a7d4-587ea0aa3e1b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY2NDE1MzgsIm5iZiI6MTcwNjY0MTIzOCwicGF0aCI6Ii8xMjc1OTE1NzIvMjc3OTcxMjA0LTZjMTE3MWRlLTk5MTEtNDI2OC1hN2Q0LTU4N2VhMGFhM2UxYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEzMFQxOTAwMzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMGRjNzllMzM2MWExMTcwMTAyNTE5ZGVhZjgxZGE1MjNjYTBkZDU2ZGFlMTRkN2Q2NjRlZmY5OTk3MTI4ZGQ2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.t_QHvjqrCFSoNVO82j0j3-kKlPb2qQ9JQB37VJAmxOY) Does anyone know how to handle the problem? Thank you in advance!

I solved this problem. The answer is updating the g++ to version 8.3.

closed 2024-06-05 by Mystic4l