unitreerobotics/unitree_legged_sdk #125: robot_interface.so: undefined symbol: PyInstanceMethod_Type

Liu-Xiaoyan97, 2023-11-08:
Now I am trying to run example_walk.py, but I got the ImportError that 'robot_interface.so: undefined symbol: PyInstanceMethod_Type '.  I confirm that I have compiled the relevant files successfully and have robot_interface.cpython-38-x86_64-linux-gnu.so (python3.6 for aarch64 used) in lib/python/amd64 and arm64. And I renamed the file in arm64 to robot_interface.so and put it in the same directory as example_walk.py. Anyone can help me? @k-okada @sktometometo @holiray @matheecs @OnlineMC 
[image]

UnInterstedLedge, 2024-05-03:
have you been able to figure this out?

Liu-Xiaoyan97, 2024-05-04:
> have you been able to figure this out?

I recommend recompiling with the version of python you're using, which will solve most of the problems. This problem has been solved, but the specific solution, I can't remember, should be to recompile.

Liu-Xiaoyan97, 2024-05-04:
Issue has been solved.

closed 2024-05-04 by Liu-Xiaoyan97

UnInterstedLedge, 2024-05-04:
i got the example_walk.cpp to work instead. so it's fine now.