unitreerobotics/unitree_legged_sdk #126: Instructions for A1 Motor Manual Replacement

Charlescai123, 2023-11-09:
Dear All,

I am currently conducting research on quadruped robots and unfortunately, the motor of my Unitree A1 got damaged because of the bugs in the running code. There're no support and maintenance sites available in North America and I quite suffered a lot from the manual replacement process for the A1 motor. Currently I was able to get the robot work as usual and made a documentation as instructions for A1 Motor Manual Replacement. This is the link for it: https://drive.google.com/file/d/1xj6-kVzLlS25_HD0-cC1vWKbUbZLqFvc/view?usp=drive_link

Hope it would bring some helps for the community.

romesco, 2023-11-11:
Thank you for this! Can you please make the drive link public? Right now it is restricted.

Charlescai123, 2023-11-11:
> Thank you for this! Can you please make the drive link public? Right now it is restricted.

Sorry for it, just make some editings. Now it should be open to public.